Ghostkeeper/SettingsGuide #3: [feature request] Enhanced synch with the sidebar

yet-another-average-joe, 2019-06-02:
When the Settings Guide gets opened, it is synch'd with the settings category. Could it automatically open the help for the item that has the focus in the sidebar ?

Ghostkeeper, 2019-06-02:
How are you opening the guide here? There are two ways to open the guide right now: Via Extensions -> Settings Guide or by right-clicking the settings.

If you right-click the settings, the setting that you happen to right-click on will get opened in the Settings Guide. So this intersects with your feature request, though maybe it could be made more clear that this is what's happening.

If you meant that it should open the setting that has focus when you click on Extensions -> Settings Guide, that can theoretically be done, though I don't know how logical that is since there is a bit of a disconnect between the application menu and the settings list. It also needs some thought into what use the welcome page has still (but that is an issue anyway).

yet-another-average-joe, 2019-06-02:
I've always been opening SettingsGuide from the menu.
Until now, I never right clicked in the SideBar !

Thanks for the information. Problem solved ! My bad...

Thanks !

Ghostkeeper, 2019-06-02:
:smile: No wonder. Even I didn't know about that feature until I first submitted this plug-in to Cura after taking over the project, which gave rise to the first bug report too: https://github.com/Ghostkeeper/SettingsGuide/issues/1

If you have an idea on how to make that feature better discoverable, let me know. Until then, I'll consider this problem solved.

closed 2019-06-02 by Ghostkeeper

yet-another-average-joe, 2019-06-02:
Yes, it is solved. Searhcing help from a contextual menu being standard, how could it be easier and more dicoverable ?!

The problem was between the seat and the mouse.