Ghostkeeper/SettingsGuide #4: Ultimaker Dark theme

yet-another-average-joe, 2019-06-02:
Ultimaker Dark makes some bitmaps hardly readable. Some cannot be read (see Colling/Fan Speed : there's some light grey text ( . It seems they are displayed in transparency mode. Could they be displayed on a white background ?

Ghostkeeper, 2019-06-02:
I tried this, but it looks rather ugly in the zoom mode I thought. I thought maybe we can make the zoom mode have a neutral grey colour background regardless of theme. My current plan is to edit all images to avoid that grey colour... But that's quite a big task.

Ghostkeeper, 2019-06-02:
You can already see how the white background looks with settings that have an animation, like Top/Bottom Line Directions.

yet-another-average-joe, 2019-06-02:
Some text is light grey : see Fan Speed :

[image]

I don't know if other images have light grey text... I will browse the contents in detail.

I barely read that.

(But maybe it's only me ; I read some documents about web page design, and I know that low contrast is better for most people reading a monitor ; on my side I need high contrast, or I get eye strain)

Ghostkeeper, 2019-06-03:
Looking again at the GIF animations currently in there it actually doesn't look as bad as I remember. I'll reconsider.

That grey text still has to go though. This is one of the reasons why I've done some work in properly rendering SVG files recently. They are much easier to edit (and cut down on the file size a bit too).

Ghostkeeper, 2019-06-09:
Implemented in version 1.3.0. All pictures now always have a white background regardless of theme.

yet-another-average-joe, 2019-06-09:
Great, SettingsGuide is now compatible with the Dark Theme !

But there are some glitches : some pictures are displayed smaller when clicked. For example Line Width or Support Pattern. This being said, as they are now resized, there's no need to click to enlarge. Resizing the dialog does it.

Ghostkeeper, 2019-06-09:
Yeah, I saw that too. It's a different issue though. It's sort of working as intended right now, but maybe we need to reduce the maximum width of single images in the image grid a bit.

closed 2019-06-09 by Ghostkeeper