Ghostkeeper/SettingsGuide #5: [Bug] Cropped bitmaps when zoomed in

yet-another-average-joe, 2019-06-02:
Hello,

When clicking a bitmap, it is displayed in a popup, but this popup is limited to the Cura Settings Guide window boudaries.

It seems to me that there was'nt such a problem with the previous version.

Also, it is displayed transparent. Could it be opaque, and also larger (maybe 3x instead of 2x) ?

Ghostkeeper, 2019-06-02:
The images grid and zoom preview has been rewritten in version 1.3 (though it looks mostly the same). In an effort to make things simpler, it is now always 2/3rd of the window size. Here's how it looks:

[image]

You think that's fine or should I make it a bit bigger? We could try 3/4.

Before [this change](https://github.com/Ghostkeeper/SettingsGuide/commit/ba410cbccaf7d7c18731132e0d0d4941711c0e89), it was always 700 by 400 pixels, which was slightly smaller than what it is now with the default window size and much smaller if you're working on a high-DPI screen (because the image didn't properly scale with DPI).

Ghostkeeper, 2019-06-02:
I've also fixed the 90% transparency of the zoom image last week, by the way: https://github.com/Ghostkeeper/SettingsGuide/commit/f4d373f63fade98baae8b5cb64092087b62e5b83

yet-another-average-joe, 2019-06-02:
I downloaded the new plugin, and copied the new files.

It seems the problem is solved now, as the window can be resized !

Therefore, it is probably no more an issue.

Thanks !

closed 2019-06-03 by Ghostkeeper