Ghostkeeper/SettingsGuide #6: [Bug Report] Maximized image not closed

yet-another-average-joe, 2019-06-03:
Hoiw to reproduce the bug :

1 - open Cura Setting Guide
2 - click an image or an animation*
3 - close Cura Settings
4 - right click ( 🥇 ) another parameter
5 - Cura Settings Guide will open the help for the new parameter, but with the previous bitmap/popup/animation overlapping the new one

When closing Cura Settings Guide, the main window should close the maximized popup/bitmap/animation

(also, the title always have (undefined) following "Cura Settings Guide")

Ghostkeeper, 2019-06-04:
Hmm, I'm unable to reproduce this. Once I open up the settings guide again it only displays the new setting. No image is zoomed.

yet-another-average-joe, 2019-06-04:
Here's an image.

I first right clicked "Infill Density" in the Sidebar, then the picture in the guide. I closed the guide without closing the zoomed image. Then I right clicked "Infill Overlap Percentage" in the Sidebar. You can see the zoomed picture for "Infill Density" overlapping the "Infill Overlap Density" one :

[image]

yet-another-average-joe, 2019-06-04:
The problem disappeared after I uninstalled/reinstalled the plugin.

I close this issue.

closed 2019-06-04 by yet-another-average-joe

Ghostkeeper, 2019-06-06:
Maybe some file hadn't been copy-pasted over it properly, or maybe you installed it initially while Cura was running.