Ghostkeeper/SettingsGuide #7: [Feature Request] Non modal window

yet-another-average-joe, 2019-06-03:
Settings Guide lives in a modal window. This makes sense when using only one monitor. On a multi head computer, it would be great to have the help on the 2nd monitor, non modal, with an automatic synchronization (parameter <-> help). I think that many people use at least 2, even with a laptop.

A workaround I've been using since 3.6 is opening a 2nd Cura instance. But there is no synch...

Ghostkeeper, 2019-06-06:
I've made the window non-modal. That makes a lot of sense. You can now keep it somewhere behind while you use Cura.

Syncing the selected setting with the setting that has focus in the main window requires some investigation. I didn't try that yet.

yet-another-average-joe, 2019-06-06:
Great !

But I'm not sure about the installation. The new feature is not available if Settings Guide is installed from the Marketplace or the .curapackage file. The install packages seem not to be up to date.

I downloaded the source files, and overwrote the previous ones with the new ones (adding SVGImage.py) in the SettingsGuide2 folder and subfolders, and letting plugin.json.in alone. Is this correct ? Last time I tinkered with Settings Guide, something went wrong ( #6 ). I don't really know what I did... Settings Guide even appeared 2 times in the Marketplace / Installed list !!! But now, it seems everything's fine.

Ghostkeeper, 2019-06-06:
Yes I haven't updated the installation on the Marketplace yet or the latest release version on Github. I also see that I forgot to add SVGImage.py to the CMake script (will do that). So the last method ("from source") is the only one that works right now until I make a new release.

Ghostkeeper, 2019-08-17:
I'm afraid that syncing the currently selected article with the article in focus in the main window is impossible. The focussed item is only known to the QML scene (it never enters the Python memory space) and the main window of the QML scene is not accessible from the SettingsGuide window.

The window is modal now though, so the important part of this ticket is done. I hope that'll be enough.

closed 2019-08-17 by Ghostkeeper