Ghostkeeper/SettingsGuide #10: Resize Viewing Size of Guide in 4.2.1

ashleycawley, 2019-09-19:
On high resolution systems it is hard to read the guide due to small font sizes and also the images look quite small on a high resolution system.

It would be good to have a simple plus or minus symbol in order to increase or decrease the viewing size, much like modern web browsers have. It would certainly make it more accessible to a wider audience.

I tried a couple of different common hotkeys used by modern browsers to increase the size of the viewing experience but to no avail.

I would like to thank you for your time and help working on this great project.

Ghostkeeper, 2019-09-23:
Since the guide is using Cura's theme, I'd expect that you'll have the same problem with Cura's interface itself? Cura's texts should read your system's DPI settings and adjust accordingly. Or maybe the problem is less pressing there because Cura's texts tend to be shorter and less of a paragraph.

Note that the images should adjust themselves based on the size of the window. You should be able to resize the window.

ashleycawley, 2019-09-23:
Thanks for the reply. No I don't have the same problem with Cura's UI in general.

You may be right about it being less pressing if all other text and labels elsewhere are shorter. If that is the case then it may be worth the guide being 2pt larger than it is presently.

The images adjust themselves in terms of location, centralising themselves sometimes but they don't scale from what I've seen. I've tried maximmised and with varying sized window, I've never seen them adapt their size though.

My resolution is 1920x1080. I have a multi-monitor setup and this screenshot below also includes a small proportion of my second monitor (just the GMail section). I have left the Github window and GMail side bar there as a comparison to show you that the text in both other windows is larger than the text in the guide.

I haven't altered any text-sizes in my OS. I'm using Windows 7 64bit btw if its any use to you.

[image]

Also in the screenshot see what I mean about the images - the guide and images are incredibly useful, thank you for producing it, but if the images scaled up to perhaps 2 or 3 times their normal scale (when the window was resized) then that would be superb. Either that or were clickable to enlarge them if required.

Thanks for your time and effort spent on this project.

Ghostkeeper, 2019-09-28:
I've made the font size somewhat bigger now.

[image]

The font size should still scale with the screen DPI settings, like it did before. But I hope that this makes it a bit easier to read with long lines.

The image thumbnails are still the same size as before, but I've also slightly increased the size of the zoomed images. I've heard before that people found those to be somewhat small, especially for diagrams. And it was a bit weird that the thumbnails could be wider than the zoomed image if the image was very wide and shallow (landscape). So that should be better in version 2.1.0.

closed 2019-09-28 by Ghostkeeper