Ghostkeeper/SettingsGuide #11: Add also the Printer setting in the parameter list

5axes, 2020-01-16:
Hello,

Is it possible to add in the parameter list also the printer settings ?
[image]

Ghostkeeper, 2020-01-17:
Yeah this is possible. I've had it in mind for a later version. Those printer settings are not visible normally though, only through the Printer Settings plug-in that you installed.

You should know though that most of those settings are meant to be descriptive. For instance, changing the printer width in Cura will not also change the printer width for your print. Usually there will not be a whole lot you can change there.

5axes, 2020-01-17:
Yes, most of the settings description are already present in the plugin. And you can have already access to the description via the right clic button.  but if if you haven't installed the plugin "Printer settings" you don't have access to these descriptions. Adding this group of parameter is of course not absolutly necessar, but it's just a little extra, like the trouble section that you added in the current developpement stage.

Ghostkeeper, 2020-04-06:
I've now completed these articles. They will be added in the 2.3 release which is due in the upcoming week in order to support Cura 4.6.

closed 2020-04-06 by Ghostkeeper