Ghostkeeper/SettingsGuide #12: [feature request] Add the possibility to manage translation of the plugin

5axes, 2020-01-16:
Get from the configuration file (cura.cfg) the language parameter in the general section.   And use this information to define the new root for the md description file in order to add the possibility to integrate translated description file.

Ghostkeeper, 2020-01-17:
This is certainly possible. It's easy for a plug-in to read the language preference from Cura. The original version of the SettingsGuide had the article files in a folder specific to English, to maybe facilitate in the future to translate the texts.

I've removed that since I probably won't ever translate everything and also keep adding new things that would then have to be translated again. This plug-in contains roughly 20 times as much text as Cura itself, and I've put in hundreds of hours into writing these articles in the first place, so it's become prohibitively expensive to translate the texts into other languages as well, something on the order of 150.000 euros. I can't afford that for just a hobby project! And even Ultimaker probably won't do that.

There is actually one fork of this repository that translates an article into French: https://github.com/ToaoF999/SettingsGuide/commits/master But it doesn't switch languages based on the preference.

5axes, 2020-01-17:
Thank's @Ghostkeeper . There is a FB Groupe  for the French native langage people using Ultimaker Cura. Some guys seems to be motivated to do the job. And there is two solution to do that the first one is just translate every MP articles and make an independant solution for this group , but on my point of view it's not a good solution and it will not grant a futur to this work. Or have the possibility for other people to integrate their own translation in the plugin ( basicaly just get the Cura Language setting, check if the file exist in a associated Language directory, if not switch to the default file) .  If this feature would be present in a next release of your GREAT plugin :) We could try to launch the work.

Goodfeat, 2020-03-09:
I also support the ability to add separate files for various localizations.

This guide is well written.
I think the community is interested in gradually adding their own translation options to this guide themselves

Ghostkeeper, 2020-03-09:
My intention is to allow incomplete translations so that not the entire guide needs to get translated at once. If a translation exists, a language button would appear (similar to Wikipedia) so that you can read a translation.

For me this is basically next on the planning. I've just finished a big feature for the SVGToolpathReader so I can spend some attention on this plug-in again.

Ghostkeeper, 2020-03-11:
I've made a translation of the Layer Height article to Dutch now to test with: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/translations/nl_NL/resolution/layer_height.md

Currently on master the language of your Cura preference is shown if a translation is available. Otherwise it will default to English.

If you want to get started, you can update your fork and start a new folder in the `resources/translations` folder, named the same as the language code (if you're unsure you can also look into the [languages list](https://github.com/Ultimaker/Cura/tree/master/resources/i18n) of Cura to find your language code). I'd prefer it if you keep the same folder structure as the English versions in the `resources/articles` folder. Don't copy the images though! Instead, put relative links to the images all the way from your translated article. Links to other articles should be relative as well. Link to a translated article if one is available, or otherwise to the English article. The intention is that the Markdown images also work passably when viewed here on Github.

My plan is to add a button so that the user can manually switch between languages as well. I will also write a chapter in the contributing guidelines to show how you can contribute to the translations. It will contain more or less the same information as in this post.

Goodfeat, 2020-03-11:
if I help translate the articles myself, how can I share the translation?

Ghostkeeper, 2020-03-11:
The normal way would be to make a pull request. This is a bit of a programmer's thing though, so if you have no experience with that you may also e-mail it to rubend at tutanota com.

Goodfeat, 2020-03-11:
I understand the essence of programming. The truth is not strong python programming language. but I think I can figure it out

Ghostkeeper, 2020-03-11:
There is no Python programming involved in the translations. Just Markdown.

Goodfeat, 2020-03-12:
Unfortunately, as I did not try, neither the master version, nor any other version can be installed using method number 4. The plugin does not appear in the lists,

Сura 4.5

can be installed either by method 1 or 2

method 3 I did not use, probably you need to install sdk or use Linux

Ghostkeeper, 2020-03-12:
4 should work. But I'll see if I can provide an intermediary build for you before the weekend so that you can test if your translations get loaded.

I really just need the .md files of your translations though. You don't really need to test it for me, though I appreciate the initiative.

5axes, 2020-03-12:
Ok tested for the french langage fr_FR, works fine, I will try to motivate some people of the French Cura3D FB group to start the translation. 
I will keep you posted

Goodfeat, 2020-03-13:
I managed to install the update. I had to replace folder "resources" + CuraSettingsGuideю.py file

Now I checked the translation of ru_RU and it works great

Goodfeat, 2020-03-13:
this is the first test article, the file is attached

Only the title is not translated
[image]

[layer_height.zip](https://github.com/Ghostkeeper/SettingsGuide/files/4327524/layer_height.zip)

5axes, 2020-03-13:
Take care also to not duplicate the images otherwise with alreay 20mb , if everyone duplicates the images folder the size of this plugin will become bigger than Cura itself :)
so every Path to the image must be change from `../images/`  to `../../../articles/images/`

5axes, 2020-03-13:
Oups !!! Sorry I haven't open your file before to write this comment .. everything is perfect

Goodfeat, 2020-03-13:
I seem to figure out how to change the path.

is there any way to reload the plugin guide without restarting the сura program?

сura launch takes a long time to test the translation of each translated article

Ghostkeeper, 2020-03-13:
I'll look into the page title. I'd think that it would take the translated setting name from whatever language the user has selected in Cura (it should already do that in the current version on the Marketplace).

There is no way to reload the plug-in without restarting Cura at the moment. After the start-up the plug-in will lazily load the articles in the background so that they can be displayed more quickly and displayed in the tooltips of the settings if that preference is enabled. This loading takes a few seconds only, so a few seconds after starting Cura the files are loaded in and won't re-load until Cura is restarted.

Goodfeat, 2020-03-15:
Hi, I did a full translation of the permission folder

ru_RU
[resolution.zip](https://github.com/Ghostkeeper/SettingsGuide/files/4334167/resolution.zip)

Ghostkeeper, 2020-03-15:
I solved the title initially by taking the translated setting name and putting that up as the title of the article: https://github.com/Ghostkeeper/SettingsGuide/commit/2aeb56691b6cc1f7b60022337b27f904afd64dcb

However I realised that there will be no translations for articles that are not settings. So instead I opted to put the title in the article itself. This also solves the issue that there is no title when viewing the [articles on Github](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/resolution/layer_height.md) using Github's Markdown rendering.

Thank you for those translations. I'll add them to the repository and add the Russian setting names to them.

5axes, 2020-03-15:
Did you also validate the French translation via the pull request ?

And as I understand we must add the article Title in the article itself ?

Ghostkeeper, 2020-03-15:
Ah, I didn't see that yet! I have yet to go through my e-mail. I'll look at that tonight too.

Ghostkeeper, 2020-03-15:
The title needs to be in the article. If you can do that would be nice. Otherwise I'll copy-paste them from the translation files.

5axes, 2020-03-15:
Ok we will update the articles in the next revision of the text. But it will takes times to modify every article.

Ghostkeeper, 2020-03-15:
Actually, let me do that. It's a brain-killing task for you guys but I can probably automate this with a Python script. It's the least I can do for you guys. That translation is huge!

Ghostkeeper, 2020-03-15:
Oh I see you've already begun. Nevermind. I'll give it a moment.

5axes, 2020-03-15:
I think i'ts better to do that manualy , because in the same time we can check the txt. We have an issue with the link on the other arcticle for wome of then to get the right result we must add the './' characters for the file in the same directory.

Ghostkeeper, 2020-03-16:
I've now finished the implementation of this. This is how it looks:
[image]

So if there are translations available for the current article, this button will appear that when clicked, will show a drop-down allowing you to select from the available languages. Clicking on one of those will change a preference in the background that determines which language gets shown, so that the next time you load an article it will stay in that language, and this will be remembered if you restart Cura too.

Like mentioned in this topic before, translations can be added according to these instructions in the contributing page: https://github.com/Ghostkeeper/SettingsGuide/blob/master/contributing.md#translating Once the .md files are added to the folder, the plug-in will automatically show the translations in the interface.

I consider this issue fixed then. The French translations by 5axes and that Facebook group will be discussed in his [pull request](https://github.com/Ghostkeeper/SettingsGuide/pull/16). I need to go to bed now so it'll have to wait until later (probably tomorrow evening).

closed 2020-03-16 by Ghostkeeper

Goodfeat, 2020-03-16:
Hi, I thought to write it here, and not create a new topic. I found something, inaccuracies relate to translation management, as well as some of the links from the original articles lead nowhere.
In my translation, I took this moment into account. It turns out that in the original translation there are no links to folders from where to download articles.

[image]
[image]

this is corrected by adding in front of the address of the folder where it comes from
[image]

Another drawback that I noticed is that when you turn on the checkmark "Show articles in setting tooltips", articles in the drop-down window still do not translate.

[image]

Another point with pop-ups with tips is that the articles in the current form do not fit. They do not scale and do not fit the monitor.
[image]

Maybe if there is no way out of this situation, it is better to make a button that will send to the article in the guide? Well, or as an alternative, you can make the "read more" button, and in the pop-up window show only part of the text.

5axes, 2020-03-17:
Ok French translation  finished and tested ...
In the futur there will be certainly some modification and new image but it's ready for a first release.

Ghostkeeper, 2020-03-17:
> It turns out that in the original translation there are no links to folders from where to download articles.

The links are relative, so just `bottom_thickness.md` should work fine as long as there is a file called `bottom_thickness.md` in the same directory. For my the links work correctly in the Top/Bottom Thickness article.

> Another drawback that I noticed is that when you turn on the checkmark "Show articles in setting tooltips", articles in the drop-down window still do not translate.

Ah, yes! I missed this I think. I'll see that it gets fixed.

> Another point with pop-ups with tips is that the articles in the current form do not fit. They do not scale and do not fit the monitor.

Normally the tooltips should be much wider than what your screenshot is showing:
[image]
However the problem you mention still exists and it's the major reason why I didn't enable this preference by default yet. You can't scroll through these tooltips since they disappear once you hover them. I can't really edit them from within this plug-in as far as I know. There is a button to "show more", really: you can right-click the setting and press "Settings Guide". This is the best solution I can do at the moment since this was meant to be expansible by plug-ins. The rest of the interface wasn't.

5axes, 2020-03-17:
That's not tottaly exact concerning the tooltip and it's translation.  If you load the article in the regular SettingsGuide windows firt .. then the text in the tooltip are in the correct langage. But if you directly display the toolwith then the text is in the default langage

Ghostkeeper, 2020-03-17:
Yeah, that's what I missed, I said. I'm on it.

Ghostkeeper, 2020-03-17:
Seems to work all right now, but I need to debug it a bit more what happens when the user switches the language without restarting.
[image]

Ghostkeeper, 2020-03-18:
@Goodfeat, this issue was to me about building a system that allows for the translations, which I consider now completed. Adding the translations themselves can still be done of course. So feel free to post them here or e-mail them to me (using my e-mail as listed on the [Marketplace](https://marketplace.ultimaker.com/app/cura/plugins/Ghostkeeper)) or to submit them in a pull request as 5axes did.

Goodfeat, 2020-03-18:
I am currently translating. but I do the translation alone, and therefore will take a little longer.

I will publish translations in archives. I will publish each folder in this thread, and will send it to you by e-mail.

Additionally, I forked the project, but I don’t know yet how it works.

Ghostkeeper, 2020-03-18:
> Additionally, I forked the project, but I don’t know yet how it works.

Forking is on the path of making a pull request:

1. Fork the project.
2. Clone your fork so that you have the files locally on your computer.
3. Make modifications to the files (e.g. add these translation files).
4. Commit your changes with Git.
5. Push the commit(s), so that they are online on [your fork](https://github.com/Goodfeat/SettingsGuide).
6. Visit the fork on Github and a notification will appear to ask if you want to make a pull request.
7. Ghostkeeper reviews the pull request and merges it if everything is okay.

But all of this is unnecessary for now. Putting it in this thread or e-mailing it works just as well. I'll author the commits in your name.

Goodfeat, 2020-03-18:
Thank you for information, I will finish translating another folder soon

Goodfeat, 2020-03-22:
[shell.zip](https://github.com/Ghostkeeper/SettingsGuide/files/4365157/shell.zip)

I finished translating another folder. I checked the links, everything seems to work as expected

Ghostkeeper, 2020-03-24:
I've tested your new translations and they work perfectly. Good work!

Goodfeat, 2020-03-25:
Thank you, I continue to translate further