Ghostkeeper/SettingsGuide #13: Plugin causes an error when starting Cura

ChrisTerBeke, 2020-01-31:
```
2020-01-31 16:35:41,787 - ERROR - [MainThread] UM.Settings.ContainerStack.getConfigurationTypeFromSerialized [342]: Could not get configuration type: 'type'
```

When printing the serialized data:

```
[general]
version = 4
name = settings_guide_stack
id = settings_guide_stack

[metadata]

[containers]
0 = empty
1 = empty
2 = empty
3 = empty
4 = empty
5 = empty
6 = settings_guide_stack_settings #2
7 = settings_guide_definitions
```

Ghostkeeper, 2020-02-04:
Yeah, it's been doing this for a while. Something to do with the initialisation order since it's loading a definition file before the rest of the resources have been loaded, I think. It doesn't seem to hurt anything, but it's a bug nonetheless.

closed 2020-03-17 by Ghostkeeper

Ghostkeeper, 2020-03-18:
The error is most likely in Cura itself somewhere. It was raised upon registering the container stack with the settings guide extra settings. This stack is not a GlobalStack or an ExtruderStack but simply a ContainerStack.

Because this ContainerStack does not have any extruders, a very old hack was triggered in Cura that was introduced to correctly load container stacks from before Cura 2.6, where single-extruder printers didn't have an extruder stack. Instead of properly transforming this in the version upgrade, this transformation is still done to this day on every stack that Cura loads (but usually earlies-out since everything has an extruder stack).

The transformation would serialise the ContainerStack and then deserialise it again in a GlobalStack instance, in order to convert it. The serialising somehow doesn't output the `type` metadata which is expected by the deserialising. I don't know why, but it's out of our power here (and doesn't seem to hurt Cura, especially since it dropped official support for loading configurations from before 3.6).

However it turns out that the ContainerStack created here was not used by the Settings Guide any more since I made the performance optimisations in version 2.0 of this plug-in. I removed it and now no error is raised any more.

commit de7fc18f (2020-03-17): Don't store definitions inside a stack

This stack is unnecessary, since we only need to list the definitions themselves, not the stack.

The stack was also causing errors since it didn't have any extruders and thus Cura was viewing it as a pre-2.3 container stack that needed to be transformed. This was not implemented very nicely so it's still being executed on every stack to date. This was causing bug #13. This commit fixes #13.