Ghostkeeper/SettingsGuide #14: SettingsGuide not compatible with 4.5 Beta ?

5axes, 2020-02-12:
Just instaled the 4.5 beta release of Cura, the SettingsGuide Plugin is not available in the cura menu.

Ghostkeeper, 2020-02-13:
Yeah, I know. The guide was using the Cura API not as it was supposed to. I caused this myself when I fixed a bug in Cura that now prohibits the use of the API in that way. I've fixed the guide here: https://github.com/Ghostkeeper/SettingsGuide/commit/5e80996ae793a6e70efa7a18ee41effb1a30895e

I'm planning to make a new release before the 4.5 stable version hits. It also contains articles for the new settings then. Until then, here is a preview version that will work with 4.5:
[SettingsGuide2.2-preview.zip](https://github.com/Ghostkeeper/SettingsGuide/files/4197014/SettingsGuide2.2-preview.zip)

You drag and drop the .curapackage into the Cura window to install it.

closed 2020-02-13 by Ghostkeeper

5axes, 2020-02-13:
Thank's @Ghostkeeper works fine for me. Thank you again for your very nice plugin