Ghostkeeper/SettingsGuide #18: [Feature Request] Possibility to add  dependance parameter in the regular windows

5axes, 2020-03-18:
Is it possible to have also in the regular Windows of the SettingsGuide the  dependance parameter as we have in the tooltip.
[image]

Ghostkeeper, 2020-03-18:
These dependencies can be different from printer to printer, so I'd rather not include that in the Settings Guide.

closed 2020-03-18 by Ghostkeeper

5axes, 2020-03-18:
it's not automaticaly generated ?

Ghostkeeper, 2020-03-18:
It is. It's generated by looking at the formulas in the [printer definition](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/ultimaker3.def.json). But it switches every time you switch printers. That's not really expected of a reference like the Settings Guide.