Ghostkeeper/SettingsGuide #32: New TroubleShooting article suggestion

5axes, 2020-08-25:
New article suggestion in the troubleshooting section 
[image]

**Layer Separation and Splitting**

- Print temperature is too low
- Layer height is too large
- Print Speed is too large

Sophist-UK, 2020-08-25:
I like the idea of increasing the troubleshooting section.

My proposal for the text for this would be:

**Layer Separation and Splitting**

Caused by weak layer adhesion combined with warping due to cooling.

Possible causes:

* Print temperature too low
* Layer height too large
* Print speed is too high
* Extrusion speed = Layer height * Print speed. So high combination may result in filament being extruded at slightly lower temperature or not fully molten centre or possibly slight under extrusion.
* Minor under extrusion from other causes
* Ambient temperature too low
* Filament type has a high level of warping

Possible solutions:

* Increase hot-end temperature by 1C-2C
* Reduce layer height
* Reduce print speed 
* Check extruder calibration
* Increase ambient temperature with an enclosure or room heater
* Increase line width to > 100% (< 150%) of nozzle size (to increase layer squish) - see YourTube CNC Kitchen
* Change to a different filament type less prone to warping

Ghostkeeper, 2020-08-27:
When I made the troubleshooting articles I had a fairly long list of print failures. About 2/3rds of them made it in. Mostly because it was cut off by the deadline of the new Cura release. I plan on making the rest as well, which includes layer separation. I made the most important ones and easy-to-reproduce ones first.

Layer splitting is kinda hard to reproduce on purpose. I've been using PLA for the demo photos so far so maybe using ABS would make this easier.

Thanks for the suggestion and the hints on the causes and solutions! I'll also do some research online of course, but more causes+solutions are always welcome.
Another frequent cause of this that I've seen is when using multiple materials that don't stick well together.

Sophist-UK, 2020-08-27:
It would be a lot of work, but a flowchart of diagnostic actions and decisions would be an even better solution i.e.

1. Check your extruder calibration + link to instructions - for bowden extruders and direct extruders
2. Check your hot-end isn't clogged + link to instructions for bowden extruders on how to hand feed filament and how to unclog
3. Check your hot-end maximum extrusion capacity + link to instructions
4. Calculate maximum speed based on extrusion capacity and line width and layer height + link to calculator
5. Calculate your warping + link to a (to be invented I think) warp calculator depending on filament type, object size, and ambient temperature 
etc.
etc.

I have a technical background and know how to experimentally/scientifically diagnose issues - but many hobbyists and some professional makers are not and cannot and would appreciate step by step instructions on fixing issues.

Ghostkeeper, 2020-08-27:
The format of these articles doesn't easily lend itself to such a flowchart right now. I've tried to order the possible causes by likelihood for the most part.

Sophist-UK, 2020-08-27:
Yes - a graphics flowchart would be ideal, but I suspect that a numbered list might be as good.

printingotb, 2020-11-06:
Too long retractions is another cause for this that I see people have problem with every now and then, particularly for all metal hotends. 
When there are many retraction in a row, under-extrusion and layer splitting can happen in those areas.

Ghostkeeper, 2021-09-04:
What do you think of this article?
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/layer_splitting.md

The image leaves something to be desired still. I tried and failed to produce something like the [Geeetech article](https://www.geeetech.com/blog/2018/02/troubleshooting-stop-your-print-layers-from-splitting/), so I resorted to causing layer splitting by reducing the contact area between layers: Simply printing with 1 wall.
But I like how the article itself turned out.

5axes, 2021-09-06:
Maybe it will be necessary to modify the article [magic_spiralize.md](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/magic_spiralize.md) which is linked on [warping.md](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/warping.md) to make a new link to this new article?

closed 2021-09-06 by 5axes