Ghostkeeper/SettingsGuide #34: Settings Guide Window gets hidden.

KenLucke, 2020-09-28:
If you open the Cura Settings Guide (BEST CURA PLUGIN EVER!), then click in the main window, the guide gets hidden behind the main window and won't automatically come forward.  This may confuse some (it did me for a while) because it appears as if the window has closed, but it will not reopen by reselecting the guide, as you would expect it to if it had closed.

**Application version**
All tested, 4.5.* through 4.7.1

**Platform**
Mac OSX Catalina (10.15.6)

**Printer**
Irrelevant

**Reproduction steps**
1. Open Cura Settings Guide
2. Click in main window
3. ::Poof::

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Settings Guide window gets hidden by main window, with no way to bring it forward without minimizing or moving the main window.  No "Windows" menu exists (I understand why) to choose which window is front-most, and reselecting "Cura Settings Guide" or right-clicking on a setting fails to bring the window forward.

**Expected results**
The window should either refuse to be hidden (as with the Marketplace window, which just throws a beep at you if you click outside of it), or, 1) there should be a "Windows" menu which lists open windows to select from (In understand why this is not there - there are usually no multiple windows in Cura - this menu could be set to only appear if there is a second window of some sort), or, 2) reselecting "Cura Settings Guide" from either the Extension menu or via right click on a setting should bring the window forward again.

**Project file**
N/A

**Log file**
N/A

**Additional information**
N/A

Ghostkeeper, 2020-09-29:
Yeah, it's kind of on purpose. The Settings Guide used to be placed in a modal window so that it was always in front. It's now on a normal window so that you can use Cura and place the Settings Guide in the background. I kind of like it better this way.

Perhaps as a fix I could move the existing window to the front when you re-open the window while it's already open. That would be logical.

KenLucke, 2020-09-29:
> Yeah, it's kind of on purpose. The Settings Guide used to be placed in a modal window so that it was always in front. It's now on a normal window so that you can use Cura and place the Settings [...]
That would be perfect.  And/Or add a temporary “Windows” menu item.

Ghostkeeper, 2020-10-17:
The window will be "activated" in the next version if it was already open. This brings it to the foreground and also puts the focus on the window again (so e.g. the arrow keys will scroll through the list of articles again).

closed 2020-10-17 by Ghostkeeper

KenLucke, 2020-10-17:
> The window will be "activated" in the next version if it was already open. This brings it to the foreground and also puts the focus on the window again (so e.g. the arrow keys will scroll through the list of articles again).

Beautiful.  Thank you, and, as I constantly tell people, "BEST CURA PLUGIN EVER"

Ghostkeeper, 2020-10-17:
Well thanks! I'm proud and glad you like it.

I also implemented a preference in the preferences page now to allow users to keep the dialogue modal if they prefer. There is as far as I know no option in Qt to keep the dialogue in the foreground but still allow interacting with the main Cura window.

KenLucke, 2020-10-17:
> Well thanks! I'm proud and glad you like it.

I've learned so much about tweaking my settings just by going through the guide feature by feature, even when not using them.  The graphics depicting the way things are affected are particularly beneficial.  I constantly wonder why it's not a built-in part of Cura, so more people would find it (surprising how many don't even now about the marketplace tab, let alone the guide.  I can't conceive of not exploring everything clickable in a program, but, I guess, that comes from doing previous alpha/beta testing and programming myself [long time ago in a galaxy far, far away]).

Ghostkeeper, 2020-10-19:
> I constantly wonder why it's not a built-in part of Cura, so more people would find it

It was considered, and still not completely off the table. One major concern is that we'd then want to have it translated in all of Ultimaker's supported languages as well (which is a requirement from Ultimaker's resellers). The Settings Guide contains about 750kB of pure text (in Markdown format). This is an order of magnitude more than Cura itself, and on par with the first two books of The Lord of the Rings. It would be pretty expensive, especially being a technical text.
It's not going to happen any time soon I think. Especially while Ultimaker can still assume that I'd maintain this in my free time.

KenLucke, 2020-10-19:
> It was considered, and still not completely off the table. One major concern is that we'd then want to have it translated in all of Ultimaker's supported languages as well (which is a requirement [...]

Well, that certainly makes sense, I had not thought about that aspect.  That would be a huge undertaking, even using a bunch of volunteers (translators, checkers, verifiers, etc.).  750K is a dribble in a bucket in today's bloatware world (I come from a time where you had 4K [or fewer] of memory to work with, and if you could shave 1 byte off of someone else's "optimized" code, you were a programming God - but then, originally, we programmed with a bank of switches to set each bit in the byte, one byte at a time, and if you screwed up, you had to start all over.  Punch cards were a Godsend, until you spilled the tray of them  :) ), but the task of the probably dozens of languages to translate would be daunting,

commit dd5f2418 (2020-10-17): Bring focus to dialogue when opening it when it's already open

This should also bring the window to the foreground if it wasn't in the foreground yet.

Fixes #34.