Ghostkeeper/SettingsGuide #38: Settings guide blanked out words

printingotb, 2020-11-02:
1. Enable "show articles in settings tooltips"
2. Open Settings visibility in preferences, hover the cursor over settings.

[image]

Side-note: the text under the left picture of "Connect Top/bottom Polygons says "Naive concentric pattern", should it be "native"?
Side-note2: The text under "minimum support area" Starts with "The raison d'être for", should be "The reason for"?

Ghostkeeper, 2020-11-03:
Thanks for the report! This settings visibility menu is something I hadn't thought about. I'm not sure how much I can do about it, since the setting tooltip is system-styled and not handled by Cura's theme.

However hyperlinks don't work in the main settings tooltip either so maybe it's a solution to filter them out of the description anyway. That would work.

Side-note 1: It's not a typo. I meant to say "naive", as in the simple version where they are not connected. Both are technically "native" to Cura, but maybe it's better to rephrase that to something like "concentric pattern has separate loops" or something that indicates what the difference is.
Side-note 2: "Raison d'être" is an expression meaning "The reason for existence". We shouldn't use complicated words though so it's better to use simpler words there.

Ghostkeeper, 2020-11-03:
One consistent problem with the Settings Guide is that people have trouble finding the guide after it's been installed. I've been planning to fix that by enabling the preference by default, to display the guide in the tooltips. I've been giving it some time before I applied that, precisely to filter out bugs like this...

printingotb, 2020-11-03:
I read almost the whole guide now when I installed the beta and I must say that it is real treasure when learning about new settings and understanding old ones better as well. I really like it.
Is it available to read on the web as well?

Could the settings guide get a button in one of the areas that I put a book on?
That way, it doesn't have to be enabled by default but still easy to find.
[image]

Ghostkeeper, 2020-11-05:
> Is it available to read on the web as well?

Yes, the articles are designed such that they also display reasonably well on Github: https://github.com/Ghostkeeper/SettingsGuide/tree/master/resources/articles This will display the content for the most recent version of Cura, even though the articles themselves contain conditional content for different versions.
The only problem is that the human-visible name, a translated text, is not available here so you have to find them by the internal key, which is sometimes quite different from the human-visible name. E.g. "Horizontal Expansion" is called `xy_offset` internally.

> Could the settings guide get a button in one of the areas that I put a book on?

While theoretically possible, this would be fairly hard to do. Those parts of the interface are not designed to be extensible, so a plug-in can't just register itself to appear there.
Parts that are extensible include the application menu, the context menu of settings, the tools on the left, the stages ("prepare", "preview" and "monitor") and their contents and the buttons next to the slice panel (e.g. post-processing scripts add a button there). There are probably a few others. But basically there needs to be a listmodel of the interface elements with a way for plug-ins to add to that list. There is no list in either of the places you indicated; it's just a few fixed elements; so there is not an easy way for me to add stuff there.
One theoretical way to implement this would be to replace an entire stage's contents. Like, basically the entire screen underneath the top bar. This is what the [Cura Sidebar GUI](https://github.com/fieldOfView/Cura-SidebarGUIPlugin) plug-in does. Seems a bit overkill and error-prone for the purpose of this plug-in though, and it won't work well with other plug-ins that do the same.

printingotb, 2020-11-05:
Thank you for that explanation! 
Yeah the sidebar plug-in is a must for me.

You have probably thought about it already but could the plug-in show a message box when installed saying something like "Right click on a setting to open the settings guide", together with a "don't show this again" check-box?

closed 2020-11-06 by Ghostkeeper

Ghostkeeper, 2020-11-06:
I've fixed the original issue in this thread here.

I've considered a lot of things to make the settings guide easier to find (once installed). It puts instructions on the installation page, but nobody's going to read that. I'm often against putting pop-ups in people's faces since I tend to get annoyed by software that does that. It's a "quick fix" for a UX problem, really. Good for when you want to ask a question in response to a user interaction, but not good for anything else.

A good user interface puts things in places where people expect them and can discover them. The settings tooltip is that place for a lot of people with the Settings Guide, and it's very easy to discover, so I think that defaulting the Settings Guide preference to display articles in the tooltips is a good solution. All of the old information is encapsulated in the articles. The tooltips are not really in the way. I just had to pilot the feature for a while to get bugs out that I didn't think of, such as what you reported here. I'm happy that I did!

commit e59b6daf (2020-11-06): Don't show hyperlinks in tooltip text

Hyperlinks don't work, and can break the layout with various system styles when the article is shown in the setting visibility menu. So let's just remove them.

Fixes #38.