Ghostkeeper/SettingsGuide #39: Descriptions improvement suggestions

printingotb, 2020-11-06:
Reword the last bit to clarify that it only stops at the end if lift head is enabled. One could interpret it as if the printer would wait regardless of the setting. I had to read that more than a few times.
[image]

Ghostkeeper, 2020-11-28:
The printer will wait regardless of the setting. If Lift Head is enabled, it will lift the print head while waiting. Otherwise it will just rest with the hot nozzle on your print.

Ghostkeeper, 2020-11-28:
Is this an improvement, you think? https://github.com/Ghostkeeper/SettingsGuide/commit/2bdfb296a24d7420e626d791e2380865570a9c69

printingotb, 2020-11-29:
Yes that seems easier to understand, at least for me. However, I can't get it to pause at the end. Printed a Ø2mm cylinder with a min layer time set to 10s and and min speed set to 10mm, it just printed on at 10mm/s without pauses. Cant find any wait commands in the gcode either.
[wait.zip](https://github.com/Ghostkeeper/SettingsGuide/files/5612018/wait.zip)

Ghostkeeper, 2020-12-11:
I looked into this and it appears that I was wrong on this. The intended use of Minimum Speed and Lift Head is badly documented so it's hard to track down what the behaviour was, but it seems that [the pause is indeed only executed if Lift Head is enabled](https://github.com/Ultimaker/CuraEngine/blob/59cb24b1d44e6f7757bd24dbd29e21fc7cb1fd03/src/LayerPlan.cpp#L1740).

I'll make the necessary corrections.

closed 2020-12-11 by Ghostkeeper

printingotb, 2020-12-11:
That's great!

commit 4b4a9f38 (2020-12-11): Correction: Print head doesn't pause if Lift Head is disabled

This was a misunderstanding of the intended behaviour on my part.

Fixes #39.