Ghostkeeper/SettingsGuide #54: SettingsGuide and Cura 4.9.1 generate a lot of log message

5axes, 2021-05-15:
The SettingsGuide plugin generate a lot of error message in the log file with the new Cura release.

Cura 4.9.1  Windows 10 .
Ie : `2021-05-15 15:37:18,619 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users/5axes/AppData/Roaming/cura/4.9/plugins/SettingsGuide2/SettingsGuide2/resources/qml/SidebarSettings/SettingItem.qml:21:2: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }`

Ghostkeeper, 2021-05-16:
Yeah, I've seen it. It's the result of the update to Qt 5.15.

Ghostkeeper, 2021-06-05:
The new syntax that is being suggested here (`function onFoo() { ... }`) is not available yet in Qt 5.10. As a result, fixing this syntax would break the Settings Guide for older versions of Cura. Since I can only keep one version of the Settings Guide in the Marketplace at a time, I'm afraid that we'll have to keep it this way.

The log messages are annoying, but at least the guide will keep functioning for older Cura versions.

The only way in which I could fix this is to duplicate the implementation of the settings list; have one version for Qt 5.10 and another for 5.15. But in that case I see having a couple of deprecated syntax lines as the lesser of two evils, even if it muddles the log. Sorry!

closed 2021-06-05 by Ghostkeeper