Ghostkeeper/SettingsGuide #56: 2.7.0 Installation failed

5axes, 2021-06-28:
Drag and Drop of the Cura package for installation doesn't install correctly the new version in Cura.

Tested on Win10 with Cura 4.9  and 4.10   with the SettingsGuide2.7.0-sdk7.0.0.curapackage  file.

hereafter the error message in cura after the restart 

[image]

Ghostkeeper, 2021-06-28:
It's a bit weird, because the plug-in ID for the Settings Guide is `SettingsGuide2`, not `SettingsGuide`. Maybe something is colliding with a version of the SettingsGuide you installed from source?

5axes, 2021-06-28:
Ok yes it was an issue comming from a previous manual installation. 
Thank you for directing me to the solution

closed 2021-06-28 by 5axes