Ghostkeeper/SettingsGuide #63: Troubleshooting article about dimensional accuracy

Ghostkeeper, 2021-09-08:
I'd like to add a troubleshooting article about imprecise prints and dimensional accuracy.

This needs to deal with the aspect of shrinkage, overextrusion and the way the material gets pushed out of the way by previously printed things. Shrinkage is an issue for big parts. Overextrusion is an issue everywhere but mainly for small parts. The way the material gets pushed out of the way by other parts is an issue in the walls.

Shrinkage should be quite obvious and refer to the shrinkage article to guide the user to fix it.

Overextrusion needs an explanation as to why it's a problem. Perhaps name a common source for overextrusion that causes dimensional inaccuracy: Wall Overlap Compensation, and flow changes when moving to a thinner outer wall.

The way material gets pushed aside needs a thorough explanation and diagrams to help the reader visualise what's going on. Essentially, this is the issue where you're printing the inner wall first, and it has all the space in the world to flow out on both sides, resulting in a wider line than intended. When the outer wall is then printed next to it, it gets pushed further out to one side, effectively doubling the inaccuracy caused by the meniscus shape of the bead. A similar issue happens when lines cross, but that is irrelevant for dimensional accuracy since lines only cross in the infill in Cura, and infill isn't on the outside of the print.

There is another common cause: Badly calibrated steppers and backlash in the gantry. Those are printer issues, nothing Cura can really do anything about. May be worth mentioning though.

This is going to be quite a difficult article to write, because there are so many aspects to it, and the results I have experience with may be quite different from what most people experience. This definitely needs a lot of research.

5axes, 2021-11-03:
Need also a special article dedicate to hole size.

Ghostkeeper, 2021-11-10:
Or at least a header.