Ghostkeeper/SettingsGuide #72: [FR] Add the name of the parameter

5axes, 2022-01-14:
Could be nice in the setting description to have the name of the Cura parameter.  Something wich could be automaticaly add to the description ?

Ghostkeeper, 2022-01-20:
The name of the setting is already at the top of the setting description. Cura adds it by itself, and the Settings Guide article contains it at the top of every article as well. Is this not what you meant?

[image]

5axes, 2022-01-20:
No not exactly : First of all this parameter is not visible in the standard View, Just in the ToolTip and personnaly I'm not sugin the ToolTip modification because the windows are on my point of view too big.  
Next the titl of the parameter is the translated parameter , I Would like to have the real Parameter Name  

IE : Infill Density => **infill_sparse_density**

And last point in the standard plugin Windows you don't have this parameter name :
[image]

So would be nice to have automaticaly something like : 
[image]

Ghostkeeper, 2022-01-24:
That internal name is meant to be internal. It's never exposed in Cura. I don't really see why it would be exposed by the Settings Guide.

We could put the name of the current article in the window header indeed. It will also be right below the window header though.

5axes, 2022-01-24:
For a developer , you're right normally he knows how to get these names .  For the basic user it's too hight level, but for someone who need to customize his Start and End Gcode and are looking for some parameter name like acceleration , speed diffrent temperaure etc   this request was for this guy.

Ghostkeeper, 2022-01-27:
For the start and end g-code it's indeed also necessary. The Settings Guide solves this by linking to the fdmprinter.def.json document online (see [here](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/machine_start_gcode.md#references-to-settings)):

> For a full list of all keys, refer to [this document](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json) in Cura's source code.

While I can see its use in specific use cases, I still think the use case is too limited to compensate for the added clutter of showing those internal names in the window title. Sorry!

closed 2022-01-27 by Ghostkeeper