Ghostkeeper/SettingsGuide #77: Settings guide screen is cut off at the bottom with no scroll capability

3DPreenter, 2022-03-29:
Hello, 
First off, thanks for taking the time to build a great add-on package! It does a great job documenting the Cura features and the images are fantastic to help explain some of the nuances of Cura. 

I'm using Cura 4.13.1 with Settings Guide 2.8.1 on Windows 10 and the issue I'm running into is being able to view the full settings guide window for a Cura feature. The settings guide window pops up normally when I mouse over a feature but for some reason, the scroll bar does not show up on the right side of the resulting settings guide screen so I'm unable to see the bottom of the guide. Further, when I mouse off of the Cura feature and over to the settings guide screen, the guide disappears. I've tried all the usual keys (up/down arrows, pgup/pgdown) and still am unable to scroll. Interestingly, when I right click on a feature and then click on the "settings guide" selection in the drop-down menu, a settings guide window pops up that does allow me to scroll up and down but it doesn't appear to be the same content.  

Am I missing a step to allow me to scroll the settings guide window or is there an issue with this? 

[image]

Ghostkeeper, 2022-04-25:
It should be possible to hover the tooltip to make it stay. You can then scroll with the scroll wheel or with the scrollbar.

Ghostkeeper, 2022-05-14:
Supposedly this is working again in Cura now.

closed 2022-05-14 by Ghostkeeper