Ghostkeeper/SettingsGuide #79: Support for 5.0 beta?

ofek256, 2022-05-05:
The plugin refuses to launch on the new 5.0.0 Cura beta release (SDK v8.0.0), when can we expect an update / workaround? Thanks!

Ghostkeeper, 2022-05-14:
Yes, I've been working the past few weeks to get it updated. I hope I can release it before the stable release, but time is against me a bit. Work has got me stressed out recently (the work on Cura itself) so I haven't been able to set my mind to work on this in my spare time too much. But that's getting better now.

The current state is that the guide works in 5.0.0, but not all articles have been updated yet by a long shot. I do have a lot of the experimentation done though, which is also necessary to write a guide like this. I'm still running the last few tests on alternating wall directions, but it's really just a matter of writing things down.

Ghostkeeper, 2022-05-19:
This is implemented in the 2.9.0 version of the Settings Guide.

closed 2022-05-19 by Ghostkeeper