Ghostkeeper/SettingsGuide #85: [New article] Cura 5.0 new articles on Walls distribution new settings

5axes, 2022-05-30:
Could be nice to have in the "troubleshooting" section a new article dedicate to the new settings of the 5.0 release compare to the 4.X parameters ( mainly on the shell section parameter and concerning the wall distribution) . The idea is to identify and resume in one page the new parameters.

Note : this article could be located in the troubleshooting section as many user  don't know how to modify their existing profile with the new settings

5axes, 2022-05-30:
In bold the parameters that could appear in this article? Note according to VersionUpgrade413to50.py

**removed settings** 
    "travel_compensate_overlapping_walls_enabled",
    "travel_compensate_overlapping_walls_0_enabled",
    "travel_compensate_overlapping_walls_x_enabled",
    "fill_perimeter_gaps",
    "filter_out_tiny_gaps",
    "wall_min_flow",
    "wall_min_flow_retract",
    "speed_equalize_flow_max"

**transformed settings**
    **"outer_inset_first"-> "inset_direction",**
    **"speed_equalize_flow_enabled"-> "speed_equalize_flow_width_factor"**

**New settings by comparison of fdmprinter.def.json" but I'm not sur to have the exhaustive list
    "**machine_scale_fan_speed_zero_to_one"**
    **"wall_transition_length"**
    **"wall_distribution_count"**
    **"wall_transition_angle"**
    **"wall_transition_filter_distance"**
    **"wall_transition_filter_deviation"**

    **"min_wall_line_width"**
    **"min_even_wall_line_width"**
        **"wall_split_middle_threshold"**
    **"min_odd_wall_line_width"**
        **"wall_add_middle_threshold"**
    **"min_feature_size"**
    **"min_bead_width"**

    "material_shrinkage_percentage_xy"
    "material_shrinkage_percentage_z"

    "skirt_brim_extruder_nr"
    "raft_base_extruder_nr"
    "raft_interface_extruder_nr"
    "raft_surface_extruder_nr"
    "raft_interface_layers"

    "meshfix_maximum_extrusion_area_deviation"

    **"material_alternate_walls"**
    "raft_remove_inside_corners"
    "raft_base_wall_count"

Ghostkeeper, 2022-06-02:
I think that towards the user, it's not nice to fill this up with articles on the history of Cura. This is a guide on how to use Cura's settings, not a reference of all information related to settings. If a user wants to know how to use a certain setting they can look up that specific article.

There is something to be said for that people want to know how to upgrade their profiles from older Cura versions, but in theory this is done automatically by Cura in its version upgrade system. The "transformed settings" you list are intended to select the settings that most closely match the settings of the previous version. For the most part, Cura keeps the setting IDs the same and tries to match the behaviour of values of new settings as well as possible to the behaviour of old settings, so in theory the upgrade routine is "do nothing and just start printing" to a user.

I think a better place for "what's new in 5.0" would be the what's new screens of Cura, the blog posts that Ultimaker puts out on its website, and other blogs/instructionals created by the community. The "troubleshooting" category is meant for when the user has a print problem and needs to solve that. There's not really a place in the Settings Guide for this.

I also don't like putting information that is inherently a temporary problem (like version upgrading) in a permanent reference like this.

Sorry! I don't think this is something I'd like to add then.

closed 2022-06-02 by Ghostkeeper