Ghostkeeper/SettingsGuide #89: guide covers work fields

rmcgraw351, 2022-06-04:
when trying to edit g-code in machine settings, the guide pops up and covers the g-code field preventing from sort of editing.

Ghostkeeper, 2022-06-12:
Yeah, this was reported to me as well via Cura's issue list.

It's a bit of a weird problem since the root cause is really that Cura doesn't deal well with setting descriptions as long as this one. The Settings Guide doesn't actually change the GUI there, it just changes the setting description. Cura itself shows the description in a tooltip when hovering the start and end g-code. I could make it not override the setting description for the start and end g-code, perhaps. But that'll also make it not show the guide in the settings list for those settings.

As a workaround for now, you can click on the text box, then move the mouse cursor away from the window. That makes the tooltip go away, so you can see what you type.

Painter602, 2022-06-15:
Found this issue when trying to update my printer's Start G-Code.
The settings guide pops up and blocks the field whenever I try to click in it.

I uninstalled SettingsGuide while masking my changes, reinstalling the SettingsGuide after.
Ghostkeeper's workaround would have worked, but I only found that when I came here to report the problem.

ImagineerNL, 2022-06-20:
Bumped into this issue as well.

@Ghostkeeper a same sort of issue exists on the infill pattern (and maybe with other 'long' fields)
They are too big and interfere with other items.
And in the infill pattern you have to quickly swipe your mouse to the popup (before it goes away) to be able to scroll

Perhaps it is an option to 'not show long guides', so you always see the guides popup, except when they are huge.
And exchanging those longer fields with either a hyperlink or a description to 'rightclick --> settingsguide' for more info.

opensourcefan, 2022-07-20:
It's a tough one; I love how easily you can get the info that you need without an action.

How about a dedicated slide out window pane? Maybe entering and exiting from the top or bottom where there is usually more free space. Or even the far left edge, that way the layout of the content wouldn't need to be altered.

By having its own dedicated area, we as users will know exactly where it will be and can keep movable windows away. 

Love this plugin though, it is amazing.

hybridview, 2023-02-09:
This one drives me nuts as well. It's nearly impossible to select a single line of GCODE in the editor to cut and move somewhere else. I would love an option to disable this specifically for fields that it completely covers.

5axes, 2023-02-10:
Just need to disable the option to display temporarily or permanently (the best from my point of view) the articles in the tooltip ( plugin preferences menu) . The time to make the changes.

[image]