Ghostkeeper/SettingsGuide #92: Erroneous Information in Enable Draft Shield

rachael7, 2022-07-10:
The settings guide has the following incorrect information under the setting "Enable Draft Shield":

`The draft shield can double as a Prime Tower. Since it's printed before the object, the printing of the draft shield is a way to get the material purged and properly flowing.`

This information is only partially correct. Yes, the draft shield for a given layer is printed before the part on that layer, but it is not the first thing printed by a newly selected extruder, so it does not serve the function of priming the extruder.  After the first layer, each extruder in a 2 extruder machine prints two layers at a time. The draft shield is indeed printed before the part on a given layer, but that extruder will have to print it's portion of the previous layer first, then the draft shield for the new layer, and finally the part on the new layer. The other extruder then comes and finishes that new layer, prints the draft shield for the subsequent layer, then prints the part on the subsequent layer. So each time there is an extruder change, the newly selected extruder has to first print a layer of the part, _before_ it prints a layer of the draft shield. That means that the draft shield does not actually prime the newly selected extruder. To illustrate, here is the order of operations for the first few layers:

Layer 2: Extruder 1 prints the bed adhesion (skirt or brim), then prints the part, then prints the draft shield. Extruder 2 prints it's portion of the bed adhesion (if applicable), then the part. On this first layer, the bed adhesion serves to prime extruder 2.
Layer 2: Extruder 2, still selected, then prints the draft shield for layer two and finally part layer 2. Extruder 1 then comes in and prints layer 2 of the part (there is no priming operation first).
Layer 3: Extruder 1, still selected, then prints the draft shield for layer three and then prints part layer 3. Extruder 2 then comes in and prints layer 3 of the part (again, no prime).
Layer 4: Extruder 2, still selected, prints draft shield layer 4, then part layer 4. Extruder 1 prints part layer 4 (no prime first).
Layer 5: Extruder 1, still selected, prints draft shield layer 5, then part layer 5. Extrude 2 prints part layer 5 (no prime first).

That pattern continues for the whole part.  After layer 1, each extruder must print its portion of the layer in progress, without any priming activity, then print the draft shield for a new layer and print its portion of that new layer.  I dearly wish there was a way for the draft shield (or ooze shield) to serve as an extruder primer, but as it stands, it does not serve that function. This is easy enough to demonstrate by simply slicing a two color file and watching the animation for each layer in the preview window.

Ghostkeeper, 2022-07-10:
I can indeed see that happening. In Cura 5.1 at least, it seems to print the draft shield with the extruder that starts the layer, which is the wrong extruder.

However this would be a bug in CuraEngine, rather than in the Settings Guide, since the Draft Shield is meant to be able to do this. I'll report this to the Cura team. All it should need is an offset somewhere to decide on which layer each extruder produces the draft shield.

The Settings Guide doesn't document bugs, so I'll leave the article as-is.

closed 2022-07-10 by Ghostkeeper

rachael7, 2022-07-10:
Thanks for this! I’m actually really glad for this result, since it means that maybe Cura will get changed to do what I hoped it would. And as long as I’m here, thank you for your work on this extension - it is invaluable!

Ghostkeeper, 2022-07-12:
Ultimaker added it on their backlog.

rachael7, 2022-07-12:
Sweet, thank you. I'm sure that's a long list, but good to know it is at least in the queue.