Ghostkeeper/SettingsGuide #93: Cura 5.x setting's search not working as it should when settingsGuide is installed

Thisismydigitalself, 2022-07-10:
Cura 5,x, Wintel 10/64bit
SettingsGuied 2.9.0.

**What's the problem:**
When searching for specific setting Cura unfolds all catagories tabs showing all settings.

**What should have happened:**
Cura should return only the settings that includes the searched keyword.

Ghostkeeper, 2022-07-10:
I'm unable to reproduce this issue. For instance, this is the result when I search for the word "draft":
[image]

I use the search function all the time myself, and I always have the Settings Guide loaded.

The search results do change by the Settings Guide though, if showing the guide in the tooltips (which is default). That's because Cura searches through both the label and the tooltip of the setting when you search for a keyword. If the Settings Guide changes the tooltip, the search results will change. Because the Settings Guide has more text, generally you'll get more search results. But for the most part I think the search results are better. Especially for keywords like "stringing" which are in the Settings Guide but not in the tooltips.

Which keyword(s) are you using to search, that you can reproduce the problem with?

Thisismydigitalself, 2022-07-10:
Hi there!
Searching "draft" returns the same results as with yours. 
Please try and search the keyword"flow".
When plug is install Cura unfolds ALL categories and shows lots of unrelated settings.
Without the plug, i get only the relevant settings:
[image]

Cheers.

Ghostkeeper, 2022-07-12:
Hmm, no I still only get settings with the word "flow" in their articles.
[image]
Are you finding any results that don't mention "flow"?

Many articles refer to flow rate, because adjusting the speed, layer height or line width adjusts the rate at which material flows through the nozzle. So you'll get quite a few results. But still only a fraction of the total settings list.

Thisismydigitalself, 2022-07-13:
Searching for "flow" and "retraction"
----------------------------------------

Results **_without_** the plug:

"Flow"

[image]

"retraction"

[image]

----------------------------------------
Results **_with plug installed_**:

"Flow"
(note how many more lines are hidden below and which i did not screenshot..)

[image]

"retraction"

[image]

Ghostkeeper, 2022-07-13:
Thanks, that looks correct then. All of the articles in your screenshot about flow mention the word "flow", and all of the articles in your screenshot about retraction mention the word "retraction" (or plural). Good to have that verified!

closed 2022-07-13 by Ghostkeeper

5axes, 2022-07-13:
You can keep the plugin et the search function. 
You just need in the option of the plugin to keep the  standard Help text , and not replace by the plugin text

5axes, 2022-07-13:
[image]

Thisismydigitalself, 2022-07-13:
@5axes 
You're the man! i did not know this even exists. and even if i did, the description does not even insinuate what it means.. 
Thank you. things looks much better now.

5axes, 2022-07-13:
@Ghostkeeper should write a new plugin the "**Plugin Guide for the Plugin Settings Guide**"  :)