Ghostkeeper/SettingsGuide #110: Looking for a new maintainer

Ghostkeeper, 2022-10-17:
The Settings Guide is currently looking for a new maintainer.

**Why?** I am no longer involved in Cura's development. While Cura development was my job at Ultimaker, it was also more than that. The Settings Guide was a labour I undertook in my free time and a source of great pride, that I did not because I was paid for it (I wasn't) but because I loved it. However being involved in Cura's community was also a source of stress for me. It's rewarding to be able to help, but it saps a lot of time and energy too. So now that I'm no longer working for Ultimaker, I'd like to also remove my involvement in the community to a great extent, and part of that is the development of the Settings Guide.

**Who?** I'm looking for someone with considerable experience in 3D printing process, and minor technical know-how. Ideally you'll be able to:
* Look at layer view to see what'll happen when you make a change.
* Understand why new settings have been implemented.
* Be able to understand the way in which screenshots are generated for the Settings Guide (which involves some abstract mark-up).
* Be able to make minor changes to the Python code should the rare bug occur, with my continued help and guidance if necessary.

**What?** When I maintained the Settings Guide, I made an update for every Cura minor release (X.Y). Almost every release features new settings, which would need a new article. I wrote about why the setting exists, what it's useful for and why you'd need to adjust it away from the default for different situations. Aside from that, old settings also sometimes change and also need to be adjusted if their behaviour changes. The guide has a mark-up construct for making content visible in different versions of Cura, so that the same Settings Guide release works for all versions of Cura down to 4.0. Part of making new articles involves making new screenshots for them. The Settings Guide has a fully automated system to do that from within Cura and to reproduce those screenshots if Cura's rendering changes, but it is quite technical to use still.
Once you are the maintainer, you'll get free reign to change your process however you want, of course. These were my standards, but you can of course adjust them to yours. I'm willing to help out with the programming side of things once in a while if there is a need for it and if you find the code difficult to understand.

**How?** Please reply here or e-mail me (rubend at tutanota dotcom) to discuss transfer of the repository to your account on Github. That will cause people to visit here on Github to be redirected to the most recent version of the repository. As for uploading the package to Ultimaker's Marketplace, I can give instructions for that if needed. You'll need to upload them on contribute.ultimaker.com. However since the Marketplace has no way to transfer a package to other accounts at this moment, it might involve a break in the user's auto-update functionality and we might have to create a whole new package that looks similar to the old one (something similar to what I did when the plug-in got transferred to me at the Cura 3.6 release). We'll have to see what works best I guess.

oliof, 2022-12-22:
Thanks for the years of supporting SettingsGuide! It was one of the common plugins I recommended to newcomers to Cura. I hope a new maintainer will volunteer soon.

ButlerOfTheSea, 2023-01-24:
I don't know if I have all the skills necessary to be the new maintainer by myself, but I would like to know more about the position. Sent you an email just a few minutes ago with the subject "Settings Guide Maintainer?" Responding to that email is definitely the best way to contact me. Hopefully we can connect soon and find a way to keep this incredible resource running!

nexus382, 2024-05-03:
I have a coder, Who we have done many projects together... He owns 9 printers, built scripts for them and everything... and i own 4 my self.... Sending email now, this can nnot die.

kvlineman, 2024-12-11:
Did you find a replacement? I would accept this responsibility in till we can find a real nerd who might do as good as you did for so long, I'm confident i could keep it afloat till then, may even become a decent enough nerd to just keep it

GregValiant, 2025-08-04:
@Ghostkeeper did you ever find someone to maintain the plugin?
There is a current post on the UM Forum that notes some problems with translations.  I'm noticing some odd behavior from the tooltip display.  It is "flashing on and off" at about 1/2 second cycle for some settings.  For others it works as it should.