Ghostkeeper/SettingsGuide #111: Settings guide preferences causes cura to crash/become unresponsive.

jnoper, 2022-10-26:
opening the preferences for the settings guide or closing the settings guide window freezes immediately (the window does not close) along with the rest of cura. nothing happens until I try to close cura and windows prompts with "program not responding". 

Attached is the cura log and 2 videos showing the error behavior. 

[cura.log](https://github.com/Ghostkeeper/SettingsGuide/files/9873031/cura.log)
https://user-images.githubusercontent.com/12039774/198137102-7c287d1c-7952-4451-a84e-effa35d63b1a.mp4
https://user-images.githubusercontent.com/12039774/198137103-84fab1ba-e11e-43b0-95cd-e4de911d5452.mp4

CURA 5.2
Settings Guide 2.9.1

Ghostkeeper, 2022-11-08:
May be a duplicate of https://github.com/Ghostkeeper/SettingsGuide/issues/80, since the symptoms are similar.

Ghostkeeper, 2022-11-08:
I'm also unable to reproduce this issue, and the log files don't say anything about it.

michalss, 2022-11-26:
This happening to me as well Win11..

Saskalex, 2023-03-11:
I have commented in #80 that this behaviour is linked to Microsoft PowerToys for me.