Ghostkeeper/SettingsGuide #119: Mouse over timeout is too short

sparky-summer, 2023-04-15:
The delay for the Settings Guide to appear when hovering with the mouse is too short and a bit annoying. Even while trying to change some values in Cura, the Guide pops up and distracts. 

Please add an option to change the mouse over delay for the guide to appear. 

Meanwhile, I will be thankful if someone tells me how to alter it manually. I tried to change all instances of "ToolTip.delay: " and "ToolTip.timeout: " in all the files I could find in the installed folder, but this did not change the hover delay. Please, help.

nickr1957, 2023-05-16:
Me too. I can't edit my startup gcode at all because the Settings Guide pops up and covers the whole windows. THere seems to be mo way to turn it off for that settings function and no way to temporarily diasbale, so I have to uninstall in order to edit startup gcode.

nickr1957, 2023-05-16:
So Ellen- that will do the trick for now.
I like the idea of a dedicated pane that can stat put or be dismissed but probably considerably more work.
Or a checkbox in Preferences/Settings to disable for hcode fields.

But these ate biggles, the guide is superb. Great work.